platformio/platformio-home #8042: Could not initialize project (`adafruit_feather_esp32s3_reversetft`)

henrygab, 2024-06-02:
```
PIO Core Call Error:

The following files/directories have been created in C:\src\simplehacks\web_ble2\web_hid2
include - Put project header files here
lib - Put project specific (private) libraries here
src - Put project source files here
platformio.ini - Project Configuration File
Resolving adafruit_feather_esp32s3_reversetft dependencies...

UnknownBoard: Unknown board ID 'adafruit_feather_esp32s3_reversetft'
```

Steps:
1. Open PlatformIO Home
2. Create new project
3. Create a new directory to store it (not default location)
4. Select the board (it autocompletes ... clearly listed as an option)
5. Ensure Arduino is the framework
6. Click "Finish"

Expected Results:
Fresh new project, ready to go.

Actual Results:
Above error message.
Directory is actually generated, and platformio.ini contains the following (w/o comments)

```
[env:adafruit_feather_esp32s3_reversetft]
platform = espressif32
board = adafruit_feather_esp32s3_reversetft
framework = arduino
```

henrygab, 2024-06-02:
`pio pkg update` from the PIO core CLI gives the same error:

```
Resolving adafruit_feather_esp32s3_reversetft dependencies...
UnknownBoard: Unknown board ID 'adafruit_feather_esp32s3_reversetft'
```

Also, the same results after adding a real-time exception for Windows' built-in `Microsoft Defender Antivirus` ...

henrygab, 2024-06-02:
This appears to be a **_very_** widespread problem:

Issue | Board
-----|-----
#8040 | `adafruit_feather_esp32s2_tft`
#8038 | `lilygo-t-display`
#8037 | `adafruit_feather_esp32s2`
#8036 | `esp32s3usbotg`
#8033 | `lilygo-t-display`
#8032 | `esp32-s3-devkitm-1`
#8026 | `esp32-s3-devkitm-1`
#8023 | `lolin_s3_pro`
#8021 | `esp32s3box`
#8019 | `vsdsquadronMini`
#8017 | `4d_systems_esp32s3_gen4_r8n16`
#8011 | `esp32-c6-devkitc-1`
#8010 | `seeed_xiao_esp32s3`
#8006 | `esp32s3usbotg`
#7999 | `4d_systems_esp32s3_gen4_r8n16`
#7997 | `lilygo-t-display-s3`
#7996 | `esp32-1732S019N`
#7991 | `esp32-s3-devkitm-1`

... and that's just from the last week or so ... there are many many more!

Is there any workaround?

In [#7997](https://github.com/platformio/platformio-home/issues/7997#issuecomment-2134080879), @pdurys indicated a potential workaround, but insufficient details to allow others to unblock themselves:
> I have found that I have manually copy boards folder to the project.

Any input on what that workaround entails?

pdurys, 2024-06-03:
Sorry for not being clear in my answer to myself ;-)

(https://github.com/Xinyuan-LilyGO/T-Display-S3) has folder named boards, which in my case copied to each affected project allow me to properly compile it. 
It looked like board name was slightly changed since I have created those about year ago.

Currently there are two LilyGo boards definitions in pio lilygo-t-display-s3 and lilygo-t-display
My projects created about year ago has reference to that board with name lilygo-t-displays3 from above mentioned repo

Eventually I have decided to remove whole pio, remove any stubborn leftovers, install it clean again and it started working as expected.

But I agree error message is not clear what actually is wrong.